badgerloop-software/podiv-altium #5: Power Supplies -- 5V Rail

rcastle1, 2018-09-19:


vkottler, 2018-09-20:
Where does the choice in IC come from? Also [here's the datasheet](http://www.analog.com/media/en/technical-documentation/data-sheets/lt8642s.pdf) for any other readers out there

vkottler, 2018-09-20:
Duplicating some of the discussion between the two issues but wondering where 8A requirement comes from on the 5V rail? (not saying it's unreasonable just wondering if we've characterized it)

[BeagleBone ref manual](https://github.com/beagleboard/beaglebone-black/wiki/System-Reference-Manual#54_Power_Management)'s blurb about power management makes it sound like it should stay < 1A draw for example

rcastle1, 2018-09-21:
I'll add in the parametric search I did.  That gave me about 10 options to look through. Looked for meeting some arbitrary input/output requirements, part availability, plus a few nice to haves in a power supply (Pgood pin, integrated caps, silent switcher). Linear has great technical support too and I've used a similar component in previous designs. 

Current requirements were pretty arbitrary, but purposely kept high as we're not sure what this will be powering in the future. Bringing down  to 4A didn't add many options. Over-powered is better than under-powered here I guess, but yeah getting power requirements from IO boards and processors should be done by @kevin asap. 

Haven't looked into 3.3V supplies much. Although if I'm the only one doing any power design I'm going to be partial to not doubling my workload (Not a direct copy from one to another, but usually changing out FB resistors and compensation components and redoing the math)

vkottler, 2018-09-21:
I notice you wrote `No need for dual outputs` for the [ADP2325](https://www.digikey.com/product-detail/en/analog-devices-inc/ADP2325ACPZ-R7/ADP2325ACPZ-R7CT-ND/3071273), but isn't "dual outputs" what we're actually looking to implement overall?

Also, any reason most of these are QFN? Personally I would almost like to make "not QFN" a requirement just so we have a fighting chance at putting our own components down

rcastle1, 2018-09-30:
Switched to using the LT8650S for 5 and 3.3V rails. Schematic capture complete

closed 2018-10-07 by rcastle1