badgerloop-software/podiv-altium #7: BeagleBone footprint + base mezzanine board geometry

vkottler, 2018-09-20:
Just a task stub, we'll need a footprint for the [BeagleBone](https://github.com/beagleboard/beaglebone-black/wiki/System-Reference-Manual#71_Expansion_Connectors) if we intend to use it and want to start thinking about the board geometries sooner rather than later

vkottler, 2018-09-24:
@rcastle1 what would be the best library to add something like this to? `connector`?

rcastle1, 2018-09-24:
I'd say connector if it's only going the 0.1" header rows.

rcastle1, 2018-10-03:
[image]

Couple of things about this configuration-- 

All external connectors on the same side and are board mount. Not sure if that's optimal, but I haven't seen a CAD model of the pod so I have no idea, but one side seems like a safe bet.

Beaglebone would be mounted with the bottom up for easy removal. center board would would not be removeable without removing side boards first. Given the sparse circuitry on this board, all components on the top side, I don't think we will have to take this board out often. 

Side boards should have enough clearance to remove external connector, undo mount screws, and slide to the left or right out of the processor-side D-sub

Case would be made of sheet metal (www.protocase design?). Ideally we'd be able to see status LEDs so a clear cover and good positioning in the pod would be great.

badgerbricker, 2018-10-18:
@rcastle1 @ezraboley are you guys working on this?

closed 2019-02-08 by rcastle1