badgerloop-software/podiv-altium #8: Backplane board gemoetry and layout

vkottler, 2018-09-20:
Just a task stub, should be re-assigned and updated w/ any commitments / decisions we make on this front

rcastle1, 2018-09-23:
For future reference we have narrowed in on these two concepts. The primary difference between them is the number of boards. 

[image]

The 2-board concept has a generalized processor and power board which plugs into module-specific IO board. This option simplifies our pinouts, reduces the number of connectors and therefore reduces board/BOM cost. 

[image]

The 3-board concept has a generalized power board, a separate processor board, and a module specific IO board. There is some added work of selecting another board-to-board connector and ensuring the two designs pinouts match, however this has the added advantage of being able to work in parallel not only for schematic capture and layout, but initial bring up of the power supplies and processor boards. For example, the processor board could have a bench power supply connected while the power supply card is being brought online. 

The key determining factor between the two options will be the number of people that are willing to commit to own a board, as collaboration on the same board across time zones may be difficult in Altium and Git. 

For either option, the enclosures will fit in an enclosure designed by the fabrication team (actual MEs). The electrical responsible engineer (RE) will work the the mechanical RE to ensure sufficient space is allocated, connector choice is suitable for assembly and disassembly, and internal harnesses are eliminated.

vkottler, 2018-09-26:
`This option simplifies our pinouts, reduces the number of connectors and therefore reduces board/BOM cost` can you show some evidence for that please @rcastle1

vkottler, 2018-09-26:
`There is some added work of selecting another board-to-board connector and ensuring the two designs pinouts match`?? "board-to-board" connector is 0.1" headers where the processor and IO boards use the same pinout, so that work is actually already done..

vkottler, 2018-09-26:
I don't think we want pneumatic-facing components on the "navigation IO" board, we want just sensing.

keving98g, 2018-09-26:
I was under the assumption that we were going to separate our braking hardware from our navigation inputs. As in put them on separate boards.

vkottler, 2018-09-26:
Yeah, that's what I am proposing

rcastle1, 2018-09-26:
The pinouts will be exactly the same? That seems like a premature assumption. Either way it’s not too much more work, but increasing connector count is non trivial. I’m still going to prefer D-sub for board-to-board connections due to the advantages I’ve listed previously, but at this point whatever.

That’s fine with pneumatic facing components. I have no idea what the physical layout of the pod will be and haven’t seen an overall block diagram of the pod which would be very helpful for the task of defining how many modules and what are their functions.

vkottler, 2018-09-26:
You don't need to make "assumptions" for things you have full control authority over, if you want the pinouts the same you can make them the same and use the same schematic component, or you can make them different and have orphaned pins or do whatever you want.

@rcastle1 if you want a BeagleBone processor+power board variation and a microcontroller processor+power board variation is that the intended approach? N possible implementations of each board?

[image]

Why is something like Ethernet being routed through something other than an RJ45? I also don't think we'd want JTAG coming out of this style of connector (just want 0.1" shrouded headers that natively connect to the programmer).

vkottler, 2018-09-26:
`I’m still going to prefer D-sub for board-to-board connections due to the advantages I’ve listed previously, but at this point whatever` we're not in disagreement that for two-boards in the same XY plane D-sub (right-angle) is a good choice. One of the architectures has that configuration (two-board) the other doesn't ("mezzanine" style, boards parallel but with a +z coordinate). If the friction of the two, 2xN shrouded headers isn't enough you can add mounting hardware to secure it (same solution used by D-sub).

rcastle1, 2018-10-10:
[image]
[image]

rcastle1, 2018-10-10:
This is the really rough layout I'm thinking for the beaglebone breakout board. For reference, this board needs to be the power passthrough to the IO board, a breakout board for the beaglebone to the IO board, and have  have CAN headers. I added in a pressure sensor circuit and will also add in IO circuitry we know we want on both HV and LV boards.

Depending on the additional circuitry, we can condense a bit on the right side. Couple things I know are not correct -- Beaglebone is floating, need to add 0.1" female headers, that IO connector is an 80 pin Misc devices part, need to replace with an appropriately sized loop-library component.

vkottler, 2018-10-10:
How many indicator LEDs?

rcastle1, 2018-10-10:
Has none at the moment. Add a 8 pin GPIO expander for that?

vkottler, 2018-10-10:
That would work, with 8 pins I think 4 for LEDs and 4 for toggle switches for configuration options would be nice

closed 2019-02-08 by rcastle1