badgerloop-software/podiv-altium #10: Board Outline and Enclosure Design

rcastle1, 2018-09-22:
1. Finalize connectors

- [ ]  Finalize connector style for board-to-board connector.
- [ ] Spec harnesses connector style for external IO and power distribution. We want to avoid internal harnessing here. 
- [ ] Reach out to protocase.com or others for sponsorship.
- [ ] Spec a layout size estimate for processor and IO boards. Get an initial size for the enclosure
- [ ] Create a .dxf file for the board outline 
- [ ] Integrate enclsure to the full pod assembly (how do we mount to the pod. Is it removable? Vibration testing)
- [ ] Spec and order all fasteners required
- [ ] Get PCB .step file after components have been placed and verify alignment, clearance with enclosure. Pay particular attention to clearances for inserting and removing boards. 
- [ ] Order or fabricate enclosure.
- [ ] Assemble enclosure
- [ ] Deploy system

vkottler, 2018-09-22:
Should we build a list of candidate connectors on here or somewhere else?

rcastle1, 2018-09-22:
Here's fine.  Here are the one's I've been looking at. (Just linking individuals, would need mates)
DSubs
* https://www.digikey.com/product-detail/en/te-connectivity-amp-connectors/6-5173277-2/A114311-ND/2203212
* https://www.digikey.com/product-detail/en/te-connectivity-amp-connectors/5787170-4/A31806-ND/807155
* https://www.digikey.com/product-detail/en/amphenol-fci/ICD26S13E4GX00LF/609-2804-ND/1090328 (May have to go 2 of these)
* https://www.digikey.com/product-detail/en/te-connectivity-amp-connectors/5747842-6/A32102-ND/808657 
* 0.1" Headers
* ~~ET60T/S~

This issue could also be split into sub- issues if necessary too.

rcastle1, 2018-09-23:
I'm still thinking we should only have two boards (processor and power on one, module specific IO on the other) that have two right angle male and female connectors on the same xy-axis. It seems like additional complexity to increase our board count. Can you elaborate on the advantages of doing a U-shaped geometry and three boards?

rcastle1, 2018-09-23:
For external harnesses we should not use D-subs or 0.1" headers. Soldering a cable harness to that type of bulkhead is very error prone and easy to short, pull out. Crimps are a lot more solid.

vkottler, 2018-09-23:
With "two right angle male and female connectors on the same xy-axis" the enclosure would be the rigid body that both boards attach two and ensure a solid connection, the procedure for getting the board(s) into the enclosure would be connecting them then fastening them both in. If a board needed to be taken out it would need to be unfastened then disconnected (nothing wrong with that, just something to consider).

The value proposition of a monolithic power+processing+etc board with N IO boards works fine. Consider that it's not possible to parallelize work on that monolithic board, one person has to own it.

I'm not proposing a U-shaped geometry anymore (see `"am having second thoughts on cards that plug into the backplane being perpendicular to it"` above). I'm proposing one base board that creates power rails and has the 0.1" shrouded header that provide power and generic connections to one or more other sets of the same header configuration.

vkottler, 2018-09-23:
I prefer not having a monolitic "power *and* processing *and* . . ." really just so that the work can be parallelized. I don't see where complexity is added by making one thing into multiple boards if we're already scoping a "common interface" with this architecture

rcastle1, 2018-09-23:
I think if we have boards on standoffs, it would be possible to unscrew one board, slide it out of the connector, and pull it out. Plus if we keep all components top side and don't have a bunch of internal harnesses, basic rework should be possible in the enclosure.  `Pay particular attention to clearances for inserting and removing boards.` 

I've worked on boards with multiple schematic designers. Layout is the harder part to collaborate on definitely. (In Cadence, you can have real time work on a layout pretty neat to watch)

keving98g, 2018-09-26:
Matt and I had met yesterday along with Matt and we had shared our initial thoughts about enclosures and set up our design goals for going with custom enclosures (along with looking nice). 

- Follow the change of decreasing/omitting internal harnessing with open/inverted enclosure design for easier testing and debugging
- Enable further customization (or partially customization) if needed between different modules.
- Decrease weight and design with full integration (putting on the pod) in mind.

Matt is working on a simple model with the ideas that we had come up with and that will be posted here.

rcastle1, 2018-09-30:
Any progress on board outline / mount hardware / connector orientation @keving98g @ matt?

closed 2019-02-08 by rcastle1