badgerloop-software/podiv-altium #16: D-Sub and 0.1" Header Library Components

rcastle1, 2018-09-26:
Make a component symbol for the following:
- [ ] [40-Position Right Angle D-sub](https://www.digikey.com/product-detail/en/te-connectivity-amp-connectors/6-5173277-2/A114311-ND/2203212`)
- [ ] [Shrouded 0.1" Header](https://www.digikey.com/product-detail/en/on-shore-technology-inc/302-S201/ED10524-ND/2794235)
- [ ] [Right angle DB-9 RCPT](https://www.digikey.com/product-detail/en/assmann-wsw-components/A-DF-09-A-KG-T2S/AE10921-ND/1241800)
- [ ] Right angle DB-9 PLUG](https://www.digikey.com/product-detail/en/assmann-wsw-components/A-DS-09-A-KG-T2S/AE10968-ND/1241804)

Make a component footprint for the following
- [ ] 40-Position Right Angle D-sub
- [ ] Shrouded 0.1" Header
- [ ] Right Angle DB-9 RCPT
- [ ] Right Angle DB-9 PLUG

nickbeckwith, 2018-09-26:
Have the 40 position right angle d-sub almost ready. Might need help getting a 3d model and footprint for it though.

vkottler, 2018-09-26:
What's that 40-Position D-sub going to be for @rcastle1

vkottler, 2018-09-26:
@nickbeckwith @rcastle1 if there's no 3D model available through ultra-librarian or linked from the part on Digi-Key we might want to choose a different part

nickbeckwith, 2018-09-27:
@vkottler @rcastle1 Okay, I can't find any models through either so I'll look around for another part if it's still necessary

badgerbricker, 2018-10-04:
working on Right angle DB-9 RCPT making footprint

closed 2019-02-08 by rcastle1