badgerloop-software/podiv-altium #29: TEMPLATE: Accelerometer

rcastle1, 2018-10-07:
### Purpose:
Acceleration measurement and velocity interpolation. 

### Circuit Description
This one's not too bad circuit wise. This basically requires a breakout to the [MTI-3](https://www.digikey.com/product-detail/en/xsens-technologies-bv/MTI-3-DK/MTI-3-DK-ND/5325385) Sensor. Actual mounting of the sensor is a little more tricky. We'd like the sensor to be mounted directly on the PCB as opposed to a short harness. Although, having the ability to adjust the sensor so it can line up with the direction of travel in the pod would be good. It's up to the RE. 

### Pod III Schematic 
[image]
Full Navigation [Schematic](https://uwmadison.box.com/s/fuog5mmjrutf7e8z87q4elr5liqnag0e)

vkottler, 2018-10-26:
I think deciding what interface to expose for use with the proposed module architecture is going to be hard to do right now, this issue can just encapsulating upstreaming a high-quality schematic symbol and footprint.

btobin86, 2018-11-04:
Pushed the schematic symbol and breakout board schematic

closed 2019-02-08 by rcastle1