badgerloop-software/podiv-altium #31: TEMPLATE: Front / Distance Sensor

rcastle1, 2018-10-07:
### Purpose:
We have to end the run within 100 ft of the end of the tube. This circuit will primarily be speccing out an affordable distance sensor that can accurately determine position within 100 ft. This is a little challenging as  we're in such an enclosed environment. The one we looked at last year was like $2000, but for this we probably have 100-200 for this sensor. 

### Circuit description
Most sensors will have a 4mA-20mA output, scaled to the total distance range of the sensor. So for this circuit you place a resistor in series with the output and measure the voltage across the resistor and scale the voltage to a current using V = I * R

badgerbricker, 2018-11-07:
# In Price Range:
## LIDAR-Lite v3HP and  V3
* v3 $129.99 40m 5V DC
     * https://www.sparkfun.com/products/14032
* v3HP $149.99 40m 5V DC faster sampling rate
     * https://www.sparkfun.com/products/14599
## RS232 50M and 100M Versions
* $47.41-$53.51 (its on alibaba though)
* Input: 23.0V-3.2V
* accuracy +/- 1m
* three wire serial interface
     * 100M: https://www.aliexpress.com/item/100M-Laser-ranging-module-digital-sensors-Distance-Measuring-serial-port-USB-to-RS232-TTL-signal/32702749857.html
     * 50M: https://www.aliexpress.com/item/50M-small-size-Laser-ranging-module-digital-sensors-Distance-Measuring-serial-port-USB-to-RS232-TTL/32761973599.html
     * on amazon: still not a lot of detail https://www.amazon.com/Distance-Serial-Sensor-Finder-Module/dp/B07G4F14RR/ref=pd_lpo_sbs_469_t_0?_encoding=UTF8&psc=1&refRID=D8KX440YFSQC2WK2ACXA

# Out of Price Range
## Parallax Lser Rangefinder 120M
* 120M I2C 5V DC
* $313.88
* Voltage Range 0-3.30V
     * https://www.digikey.com/product-detail/en/parallax-inc/28054/28054-ND/6009003
## Range Finder 1000M
* $375
* +- 0.5m(400m)
* 3V 120mA
     * https://www.amazon.com/Distance-Serial-Module-Rangefinder-Measurement/dp/B0761ZLYSN

badgerbricker, 2018-11-07:
# Information Needed
1. breaking distance of pod
     * need to know how much distance warning needed will it be able to stop within 100 feet or does the sensor need to have a longer range to trigger stopping earlier
2. speed of pod before breaking
     * check max sampling rate
3. what interfaces are avaible on the microcontrolers? Which one will it interface with?
     * I2C manditor? Analoge OK?
## Next steps
1. Pick sensor
2. Create part in Altium
     * Schematic and Footprint
3. Create circuit in Altium
4. Add as snipet

rcastle1, 2018-11-07:
For this sensor to be worthwhile, we should think about how it will be used in the control loop. My comments should be generic to whatever sensor is selected, but a specific one may alleviate some concerns or cause new ones. 

They will have a beta angle that determines how large the field of vision will be. The sensor will detect the side walls for most of the run, until we are a range of the end of the tube that the geometry works out. An excel tool with some of the math worked out could be useful. 

There are a few options for the control loop I can think of for using the sensor:
1. Ignore data from this sensor until we can detect that it's actually hitting the end of the tube. (how this is done is probably sensor dependent, I'm not sure if this is actually possible)
2. Rely on the redundancy in retro tape to give us an exact position with 100-ft resolution. Detect when we are at the last tape strip 100ft away from the end of the tube (or if we know the sensor has greater usable range, some retro set point) and then start reading the data.
3. See if we can determine that in the tube, we will be reading a constant value until we near the end of the tube, where the behavior changes, and go from there. 
4. Talk to an applications engineer from one of these sensor companies.  You might have to be persistent to get a hold of someone. 

Reasons for not including the sensor:
1. If we are already ignoring data and relying on retro tape for 100 ft resolution and the requirement is to stop within 100ft of the end, why do we need this extra sensor? 
2. If the run profile is still to start, go to a high speed, stop, then go for a low speed, do we just rely on manual control (with appropriate signal-loss fail safes)?
3. Testing the sensor would only really be possible at scale when we get to California and they let us in the tube. We've never made it before, and I'm assuming that we will have other problems to work out that week, so unless we can prove the sensor operates as we understand it on the bench and can prove that this will scale, this should be descoped. I'm all for buying one for $100 or whatever and trying it out as our advisers clearly wanted us to have one last year. 

@badgerbricker You should talk with @ezraboley and @vkottler about any sensors with a digital interface. I would also be a avoid solutions that come from websites like aliexpress.

rdaruwala, 2018-11-07:
To pitch in for the interfaces -- We should be able to deal with most. I2C/SPI/UART/Analog are all fine. 

Regarding the control loop:

From my (limited) experience with range sensors, they tend to return an "no range found" data value when they don't have data. Though I imagine the system will vary depending on the sensor. I'd suggest reading it throughout the run, should be a good redundancy to have.

Braking distance is a really important factor we need to consider. Do we have an approximate number?

badgerbricker, 2018-11-08:
The LIDAR-Lite V3 by spark fun is only $129 as mentioned above has a 4x2m radian divergence (elliptical) I would assume that is at the max measure distance of 40m but sparkfun doesnt specify. I2C compatable so it should be able to talk to the beagle bone just fine. Also its made by sparkfun so its will be good quality. I think this sensor is looking promissing but we need to know:
* Would 4x2m divergence avoid hitting the walls of the pod? /What is the radius of the pod? 
* Would 40m be enough for braking distance if we choose to rely on this sensor for position? Alternatively we could use it to validate the slow crawl to the end of the tunnel in addition to the retrorelfective strips.

I understand that it might not be necessary to have this sensor since we have to retros but it might be a nice secondary system to validate the retros Im not sure how accurate the retros have been in the past.

closed 2019-02-08 by rcastle1