badgerloop-software/podiv-altium #33: TEMPLATE: Pressure Sensor -- Line

rcastle1, 2018-10-07:
### Purpose:
Pressure sensors monitor what the pneumatics system is doing. They are critical for safety as pressure systems can explode. 

### Circuit Description:
There are two primary pressure sensor inputs
* 5V
* 24V
There are a few main types of pressure sensor outputs. 
* 4-20mA output
* 0-5V
* 0.5-4.5V

Our Pod III Circuit did a nice job of being configurable, as a lot of the time we may need to add in a sensor last minute after a design review and they are fairly difficult to source in these low quantities. I think the main improvement for this circuit will be improved connectors and to think through how we will interface with the sensor. Note that we probably don't need LV_Batt for this. 

One improvement that we could possibly make is to have a precision 5 or 24V reference here. Output voltage fluctuates with input voltage to a degree (particularly with higher frequency, lower amplitude noise) so a good twisted pair and a voltage reference would be a nice to have. 
[image]

clendening, 2018-11-01:
Snippet pushed to sensor_list

rcastle1, 2018-11-04:
this could be a good place for "connect by net" instead of "connect by wire" to be implemented. Also not sure if we should be including the breakout connectors in these circuit snippets. Also looks like the footprint for the 3-pin block is from the misc. connectors library and in an uncommitted library.

rcastle1, 2019-02-08:
Closing. Circuit complete

closed 2019-02-08 by rcastle1