badgerloop-software/podiv-altium #39: Pre-Charge Component Library Additions

rcastle1, 2018-10-16:
Comment below when you start working on one of these components and check if off once complete. 

Schematic Symbol
- [ ] CHSF-250 Fuse
- [ ] 1amp 300 V Fuse
- [ ] Contactor
- [ ] S300 YR Diode
- [ ] SPST-NC Relay
- [ ] SPST Relay
- [ ] DPDT Relay
- [ ] 1k Power Resistor

Footprint
- [ ] CHSF-250 Fuse
- [ ] 1amp 300 V Fuse
- [ ] Contactor
- [ ] S300 YR Diode
- [ ] SPST-NC Relay
- [ ] SPST Relay
- [ ] DPDT Relay
- [ ] 1k Power Resistor

Ghood08, 2018-10-16:
10/15 I'm starting to work on the s300YR Diode

rcastle1, 2018-10-16:
Looked through a few components and noticed a few things
* https://www.infineon.com/cms/en/product/transistor-diode/diode/chip-diode/sidc161d170h/#!overview This is wafer, not a component that goes on a PCB. See https://en.wikipedia.org/wiki/Die_(integrated_circuit) for more information. I've never spun my own silicon, so that would be cool, but probably not what we're looking for. It also has zero available on Digikey @Ghood08 
* BK/GLR-1 Looks like a cartridge that's normally mounted on a panel, rather than a PCB. The datasheet's not loading right now, but check out the suggested mating component. That's usually used on a metal panel for something that needs to be replaced somewhat often, without opening up the rest of the enclosure. 
* Be sure to specify the part numbers for CHSF-250 Fuse and the Contactor. 
* https://www.digikey.com/product-detail/en/te-connectivity-potter-brumfield-relays/T9GV2L14-12/PB2373-ND/8345133 Is this going to be used from the output of the motor controller? Check out the output voltage of that, I think it's only 3.3 or 5V, so you may need an optocoupler or other method to step up the voltage. Could be fine, depending on how you want to use it. 
* https://www.digikey.com/product-detail/en/kemet/EE2-12NU-L/399-10997-1-ND/4506449 I might look for a different package here. This type of lead tends to snap on fairly easily. 
* https://www.digikey.com/product-detail/en/ohmite/HS150-1KF/HS1501KF-ND/5306251 Think through how the footprint should look for this component.

yashwani2000, 2018-10-30:
Which branch do I push to? Currently I've saved changes to yashwani2000_training?

yashwani2000, 2018-10-30:
I'm starting to work on the DPDT relay and just updated the sepc'ing sheet, so could you please review the component?

rcastle1, 2019-02-08:
Components added. Closing

closed 2019-02-08 by rcastle1