badgerloop-software/podiv-altium #60: HV_IO Breakout

rcastle1, 2019-03-23:
@shelby noticed that I (stupidly) didn’t breakout the shutdown circuit lines to the DSub. Issue was resolved by connecting 18 awg wire to the underside of the PTH on the jumper to the PTH on the dsub. This is all just 12V <0.5A to drive the contactor, so this solution should be ok. All wiring is on the bottom side of the board too ;)

@Shelby can you comment with the updated pinout? 

If the board is REVed, add breakouts.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

rcastle1, 2019-03-23:
@dawiggleman ^

dawiggleman, 2019-03-24:
Yes,
Connector J4B:
Pin 13: Inertia Switch +
Pin 14: Inertia Switch -
Pin 17: Master/E-Stop Switch +
Pin 18: Master/ E-Stop Switch -

rcastle1, 2019-03-24:
Thanks. Were you able to test this out? I saw the rework today

rcastle1, 2019-03-24:
Will be resolved in REVB

rcastle1, 2019-04-22:
Board is assembled, in queue for validation. 

(this got sidetracked due to other priorities due to reveal. Aiming to have board validated by 4/24)

rcastle1, 2019-06-09:
Rev B functioning as expected.

closed 2019-06-09 by rcastle1